eegsynth/eegsynth #311: add ini filename to Patch class

stephenwhitmarsh, 2019-01-25:
This would make the main loop cleaner, and the Patch class more modular. It will improve sharing between (python) modules that can read/update/write the Patch instance with user-defined (gui) data, and save/load/read to ini file with its own functions.

robertoostenveld, 2019-01-26:
the patch class could also be connecting to redis internally (on basis of the settings in the ini), i.e. you would not see any redis related code in the modules any more.

robertoostenveld, 2019-01-26:
putting the configparser inside the patch class also makes it easy to (for now) keep on supporting the inline comments.

stephenwhitmarsh, 2019-01-26:
thats the idea!

robertoostenveld, 2020-04-13:
With the new modular structure this has more or less been addressed

closed 2020-04-13 by robertoostenveld