eegsynth/eegsynth #320: implement support for Doepfer MCV4

robertoostenveld, 2019-04-01:
see http://www.doepfer.de/mcv4.htm and the PDF on that page.

It has 4 CVs, one gate and one s-trig(?).

The CVs are programmable, but seem controlled with different MIDI messages (CV1=Pitch, CV2=Aftertouch, CV3=Volume, CV4=Modulation), where  CV3 and C4 can be multiplied with MIDI velocity. 

The way the MIDI control commands are used is not yet totally clear to me yet, but it does not seem a good match to the generic outputmidi module. Hence I propose that we test the mcv4 with the eegsynth, and if needed, that we implement a custom "outputmcv4" module for it.

stephenwhitmarsh, 2019-04-01:
agreed, was looking into it, but now don't have time this week - off for a
couple of days of climbing in Espagne. First thing next week!

robertoostenveld, 2019-04-13:
# Doepfer MIDI to CV/Gate interfaces

The [A-190-2](http://www.doepfer.de/a1902.htm) is the modular version of the [MCV4](http://www.doepfer.de/mcv4.htm). The [A-190-3](http://www.doepfer.de/a1903.htm) is very similar to the A-190-2, the main difference is the additional USB interface and the behavior of the LED. The A-190-3 is the modular version of the [Dark Link](http://www.doepfer.de/Dark_Link_e.htm). Hence, all four interfaces are controlled in the same way and have the same core functionality.

## CV1

This is controlled by MIDI note messages, 1V/Octave, 0 to +5V. The glide control (portamento) button can be used for CV1.

## CV2

This is controlled by MIDI pitch bend, approximately -2.5 to +2.5V or approximately 0 to +5V (can be selected by an internal jumper).

## CV3

This is controlled by MIDI velocity, 0 to +5V.

## CV4

This is controlled by MIDI control change messages, free assignable controller in learn mode.

robertoostenveld, 2019-04-13:
The only reason why at this moment the CV1 cannot be controlled, is that `outputmidi` has a 1-to-1 mapping of Redis messages on MIDI notes. I will see whether I can implement a 1-to-many mapping, so that one Redis message (e..g alpha power) is mapped onto any MIDI note (modulating the pitch of the note). 

I think this issue (i.e. "how to interface the INI file specification to Redis and from there to MIDI") also applies to the `sequencer` module.

commit f250889e (2019-04-15): FIX #320 - added support for mpping continuous Redis channels onto MIDI pitch, this is required for CV1
closed 2019-04-15 by robertoostenveld