eegsynth/eegsynth #327: unresolved variable

stephenwhitmarsh, 2019-06-14:
Variable 'window' is unresolved: Line 409 in lib/EEGsynth.py
@robertoostenveld : it's in your filter function ;-)

robertoostenveld, 2019-06-14:
it is in **our** filter function!

robertoostenveld, 2019-06-14:
I think it should be `b = np.zeros(1)` but that is something to test

````
    if not(highpass is None) and not(lowpass is None) and highpass>=lowpass:
        # totally blocking all signal
        print('using NULL filter', [highpass, lowpass])
        b = np.zeros(window)
        a = np.ones(1)
    elif not(lowpass is None) and (highpass is None):
        print('using lowpass filter', [highpass, lowpass])
        b = firwin(order, cutoff = lowpass, window = filtwin, pass_zero = True)
        a = np.ones(1)
    elif not(highpass is None) and (lowpass is None):
        print('using highpass filter', [highpass, lowpass])
        b = firwin(order, cutoff = highpass, window = filtwin, pass_zero = False)
        a = np.ones(1)
    elif not(highpass is None) and not(lowpass is None):
        print('using bandpass filter', [highpass, lowpass])
        b = firwin(order, cutoff = [highpass, lowpass], window = filtwin, pass_zero = False)
        a = np.ones(1)
    else:
        # no filtering at all
        print('using IDENTITY filter', [highpass, lowpass])
        b = np.ones(1)
        a = np.ones(1)
````

robertoostenveld, 2019-06-17:
I fixed it with https://github.com/eegsynth/eegsynth/commit/0cbebfcfa9236b196f32856871d1023dbe31ac87

closed 2019-06-17 by robertoostenveld