eegsynth/eegsynth #335: pyOSC is not compatible with Python3

hyruuk, 2019-08-19:
I'm trying to run an EEGsynth patch that processes Muse data from MuseLSL and send them to PureData via OSC. 
Everything seems to run well, excepted for the OSC library (pyOSC) that I am not able to install on Python 3+.

It looks like pyosc is a Python 2 library, so I tried modifying the OSC.py file that comes with it, but I have troubles to make it work (I am not very familiar with the differences between Py2 and Py3). Is there any quick and easy way to make the outputosc module work in Python 3 ?

robertoostenveld, 2019-08-19:
I have also not succeeded in installing pyOSC on python3, but not tried very hard. I should add that most development and all performances so far used python2; we only started testing on python3 recently.  Are you aware of a OSC package that is compatible both with 2 and 3?

robertoostenveld, 2019-08-19:
Perhaps https://pypi.org/project/python-osc/ can be used instead. It appears to me that this would not require too many changes to `inputosc` and `outputosc` modules. The use of OSC in midiwrapper (in EEGsynth.py) is not relevant any more, so that can be ignored; I will remove it in the near future.

If you want, please give it a try to update the `outputmidi` module using python-osc and send a PR.

tmpbci, 2019-08-20:
Hi there,

Someone made a python 3 version of the classic osc lib :

https://github.com/sergeLabo/pymultilame/tree/master/pymultilame <https://github.com/sergeLabo/pymultilame/tree/master/pymultilame>

Sam

robertoostenveld, 2019-08-20:
The OSC specific section in pymultilame seems to originate from (or be continued at) https://github.com/Qirky/pyOSC3. But it cannot be installed with pip, so my preference at this moment remains with python-osc. Help with implementing the change would be appreciated.

hyruuk, 2019-08-20:
> The OSC specific section in pymultilame seems to originate from (or be continued at) https://github.com/Qirky/pyOSC3. But it cannot be installed with pip, so my preference at this moment remains with python-osc. Help with implementing the change would be appreciated.

Am on it :) I just need to test the changes I made and I'll do a pull request this evening if everything works as intended.

hyruuk, 2019-08-21:
I modified and tested the outputosc module, it's now fine using python-osc. The inputosc module is still using pyOSC though, changing it isn't as trivial as the other (but shouldn't pose any problems either).

robertoostenveld, 2019-08-21:
This has been resolved for the `outputmidi` module with #336. The same problem will apply to `inputmidi`.

robertoostenveld, 2019-08-21:
It is now also resolved for inputmidi. In case you want to test, this is how I did it. I used redis,  launchcontrol, inputosc and outputosc modules. Using launchcontrol I injected stuff into redis. outputosc reads that, and sends it (over OSC) to inputosc, which (again, but with another key) puts it in Redis. 

I tested both python27 and 37. The code is conditional on the python version, and when using python > 3.6 (i.e. `python-osc`) it will not be as refined as with the old `OSC` implementation, which also supports multi-value arrays. But this is a start and probably deals with most common cases ...

commit dccdc824 (2019-08-21): FIX #335 - support both python2 and python3 in inputosc module
closed 2019-08-21 by robertoostenveld