eegsynth/eegsynth #354: Buffer shell script not compatible with WSL

stephenwhitmarsh, 2019-12-27:
Although buffer runs when called directly, the shell script does not work for unknown (to me) reasons:

steve@DESKTOP-NN4RFID:/mnt/c/Users/Steve/eegsynth/module/buffer$ ./buffer.sh
./../../bin/parallel: line 10: $'\r': command not found
./../../bin/parallel: line 13: $'\r': command not found
./../../bin/parallel: line 16: $'\r': command not found
./../../bin/parallel: line 37: syntax error near unexpected token `elif'
'/../../bin/parallel: line 37: `elif [ $OS = Darwin ] ; then

robertoostenveld, 2019-12-28:
The backslash r is the "return" command and dates back to typewriters. See https://en.wikipedia.org/wiki/Carriage_return

On windows, lines end in an ascii text file with \r \n as "return" and "newline", on unix only a newline is used and the return is implicit.

Git (the software) can deal with this automatically, see https://stackoverflow.com/questions/1967370/git-replacing-lf-with-crlf. In your case, I suspect you used the windows git client (which will insert CRLF) and then executed the code in WSL bash.

stephenwhitmarsh, 2019-12-28:
Sounds right! Ill pull again from WSL. Thanks :)

robertoostenveld, 2019-12-30:
I suppose this is fixed. If not, please reopen

closed 2019-12-30 by robertoostenveld