eegsynth/eegsynth #365: terminal output formatting bugs

stephenwhitmarsh, 2020-03-23:
Both preprocessing and endorphines give me issues in the terminal output.
Attached two screenshots. Seems to be caused by either formatting the path, or some type error. below a copy-paste of one section

[image]
[image]

  File "C:\Users\Steve\eegsynth\module\preprocessing\../../lib\EEGsynth.py", line 75, in format
    msg = msg % self.args
    monitor.info("preprocessed", window_new, "samples in", (time.time()-start)*1000, "ms")
Arguments: (24, 'samples in', 0.9973049163818359, 'ms')
  File "C:\Users\Steve\eegsynth\module\preprocessing\../../lib\EEGsynth.py", line 198, in info
Traceback (most recent call last):
  File "C:\Users\Steve\AppData\Local\Continuum\anaconda3\envs\EEGSynth\lib\logging\__init__.py", line 1025, in emit
    msg = self.format(record)
    return fmt.format(record)
  File "C:\Users\Steve\eegsynth\module\preprocessing\../../lib\EEGsynth.py", line 75, in format
    return colored(record.levelname, color) + ': ' + self.name + ': ' + record.getMessage()
  File "C:\Users\Steve\AppData\Local\Continuum\anaconda3\envs\EEGSynth\lib\logging\__init__.py", line 369, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "C:\Users\Steve\eegsynth\module\preprocessing\preprocessing.py", line 327, in <module>
    _loop_forever()
  File "C:\Users\Steve\eegsynth\module\preprocessing\preprocessing.py", line 310, in _loop_forever
    _loop_once()
  File "C:\Users\Steve\eegsynth\module\preprocessing\preprocessing.py", line 298, in _loop_once
    monitor.info("preprocessed", window_new, "samples in", (time.time()-start)*1000, "ms")
  File "C:\Users\Steve\eegsynth\module\preprocessing\../../lib\EEGsynth.py", line 198, in info
    self.logger.log(logging.INFO, *args)
Message: 'preprocessed'
Arguments: (24, 'samples in', 0.9992122650146484, 'ms')
INFO: preprocessing: looping with 14 iterations in 1.05258 seconds

stephenwhitmarsh, 2020-03-23:
ook Spectral

robertoostenveld, 2020-03-23:
please update to the latest version, I think that it solves these warning/error messages.

stephenwhitmarsh, 2020-03-24:
Endorphines still has the problem in latest release (haven't checked spectral yet).

robertoostenveld, 2020-03-25:
The problem was that there were still statements like

```
monitor.info(aaa, bbb)
```
which worked fine when the underlying code was still 
```
print(aaa, bbb)
```
but the recent update to use the logger instead caused this syntax to break. The logger expects a single string as input argument.

commit 5658e680 (2020-03-25): FIX #365 - fixed more terminal formatting bugs, the logger does not accept input arguments in the same format as print()
closed 2020-03-25 by robertoostenveld