eegsynth/eegsynth #368: allow wildcards for ini files

stephenwhitmarsh, 2020-03-24:
At this moment *.ini doesn't work, and one has to specify all ini files in one list.

robertoostenveld, 2020-03-24:
this seems windows specific. Can you check how `eegsynth/bin/eegsynth.py` parses the command line?

robertoostenveld, 2020-03-24:
```
(eegsynth37) C:\Users\Robert\Documents\GitHub\eegsynth\patches\loop>eegsynth *.ini
Traceback (most recent call last):
  File "c:\users\robert\anaconda3\envs\eegsynth37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\robert\anaconda3\envs\eegsynth37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Robert\Anaconda3\envs\eegsynth37\Scripts\eegsynth.exe\__main__.py", line 9, in <module>
  File "c:\users\robert\anaconda3\envs\eegsynth37\lib\site-packages\eegsynth\bin\eegsynth.py", line 73, in _main
    raise RuntimeError('incorrect command line argument ' + file_or_dir)
RuntimeError: incorrect command line argument *.ini
```

confirmed, this applies to the Anaconda Prompt.

robertoostenveld, 2020-03-24:
it is not specific to python, also applies to the built-in dir command

```
(eegsynth37) C:\Users\Robert\Documents\GitHub\eegsynth\patches\loop>dir *.ini
 Volume in drive C is Windows
 Volume Serial Number is 8A45-9411

 Directory of C:\Users\Robert\Documents\GitHub\eegsynth\patches\loop

File Not Found
```

robertoostenveld, 2020-03-24:
https://stackoverflow.com/questions/12501761/passing-multiple-files-with-asterisk-to-python-shell-in-windows

commit 85ab7d15 (2020-03-24): FIX #368 - use glob on windows command line arguments
closed 2020-03-24 by robertoostenveld