eegsynth/eegsynth #372: Playbacksignal crash without error

hyruuk, 2020-04-02:
I'm having troubles to use the playbacksignal module on the most recent update of the repo. 
It doesn't seem to return any error message, just crashes after printing the debugging info : 

`(cbc) hyruuk@hyruuk-laptop:~/GitHub/eegsynth/patches/cbc$ python ../../module/playbacksignal/playbacksignal.py                -i playbacksignal.ini

##############################################################################
# This software is part of the EEGsynth, see <http://www.eegsynth.org>.
#
# Copyright (C) 2017-2020 EEGsynth project
#
# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.
#
# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with this program.  If not, see <http://www.gnu.org/licenses/>.
##############################################################################

Press Ctrl-C to stop this module.
        
INFO: playbacksignal: Reading data from ../../datafiles/recordsignal_2019.09.08_16.07.07.edf
SUCCESS: playbacksignal: Trying to connect to buffer on localhost:1972 ...
SUCCESS: playbacksignal: Connected to FieldTrip buffer
INFO: playbacksignal: reading channel 0
INFO: playbacksignal: reading channel 1
INFO: playbacksignal: reading channel 2
INFO: playbacksignal: reading channel 3
INFO: playbacksignal: reading channel 4
DEBUG: playbacksignal: nChannels = 5
DEBUG: playbacksignal: nSamples = 21760
DEBUG: playbacksignal: fSample = 256.0
DEBUG: playbacksignal: labels = ['0', '1', '2', '3', '4']
`

Why is that ? The module worked fine before, I didn't change the datafile nor the call to the module in my patch. I guess an update broke it but I can't figure out which.

robertoostenveld, 2020-04-02:
sorry about this. The refactoring/reorganizing of the code in the new modular (in the software sense, not in the synth sense) organization was quite a job and apparently introduced some glitches like this.

Hmm, I think you not seeing the cause of the error is due to 

```
if __name__ == '__main__':
    _setup()
    _start()
    try:
        _loop_forever()
    except:
        _stop()
```
which catches a bug silently, and then gracefully stops. Please change it into

```
if __name__ == '__main__':
    _setup()
    _start()
    _loop_forever()
```

to make the error visible. 

Note to myself: the `try-catch` statement should be explicit about what the allowed exceptions are, e.g. `SystemExit`, `KeyboardInterrupt` and `RuntimeError`. Anything else should be explicitly dealt with as programming error.

hyruuk, 2020-04-02:
Thank you very much, I got the module working again ! I'm creating a PR to integrate my correction.

closed 2020-04-02 by hyruuk