eegsynth/eegsynth #374: pip install fails on macOS 10.9.5 and Anaconda 5.1

robertoostenveld, 2020-04-04:
I detected this on the old computer from Per

```
Building wheels for collected packages: zmq, python-osc, paho-mqtt, termcolor, sklearn
  Building wheel for zmq (setup.py) ... error
  ERROR: Command errored out with exit status -5:
   command: /Users/perhuttner/anaconda3/envs/eegsynth/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/sh/xsq9cpw572d9fywbp9l0j35c0000gn/T/pip-install-gfo0t1eo/zmq/setup.py'"'"'; __file__='"'"'/private/var/folders/sh/xsq9cpw572d9fywbp9l0j35c0000gn/T/pip-install-gfo0t1eo/zmq/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/sh/xsq9cpw572d9fywbp9l0j35c0000gn/T/pip-wheel-uufrtnva
       cwd: /private/var/folders/sh/xsq9cpw572d9fywbp9l0j35c0000gn/T/pip-install-gfo0t1eo/zmq/
  Complete output (23 lines):
  You probably meant 'pyzmq' not 'zmq'
  running bdist_wheel
  running build
  installing to build/bdist.macosx-10.9-x86_64/wheel
  running install
  running install_egg_info
  running egg_info
  writing zmq.egg-info/PKG-INFO
  writing dependency_links to zmq.egg-info/dependency_links.txt
  writing requirements to zmq.egg-info/requires.txt
  writing top-level names to zmq.egg-info/top_level.txt
  reading manifest file 'zmq.egg-info/SOURCES.txt'
  writing manifest file 'zmq.egg-info/SOURCES.txt'
  Copying zmq.egg-info to build/bdist.macosx-10.9-x86_64/wheel/zmq-0.0.0-py3.7.egg-info
  running install_scripts
  dyld: lazy symbol binding failed: Symbol not found: _fdopendir$INODE64
    Referenced from: /Users/perhuttner/anaconda3/envs/eegsynth/bin/python
    Expected in: /usr/lib/libSystem.B.dylib
  
  dyld: Symbol not found: _fdopendir$INODE64
    Referenced from: /Users/perhuttner/anaconda3/envs/eegsynth/bin/python
    Expected in: /usr/lib/libSystem.B.dylibЀ
```

robertoostenveld, 2020-04-04:
I closed the terminal, opened it again and tried `pip install pyzmq`, and then it told me it was already installed. When I tried `pip install eegsynth` again, I got this

```
Building wheels for collected packages: paho-mqtt, sklearn, zmq, termcolor, python-osc
  Building wheel for paho-mqtt (setup.py) ... error
  ERROR: Command errored out with exit status -5:
   command: /Users/perhuttner/anaconda3/envs/eegsynth/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/sh/xsq9cpw572d9fywbp9l0j35c0000gn/T/pip-install-fss2zby2/paho-mqtt/setup.py'"'"'; __file__='"'"'/private/var/folders/sh/xsq9cpw572d9fywbp9l0j35c0000gn/T/pip-install-fss2zby2/paho-mqtt/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/sh/xsq9cpw572d9fywbp9l0j35c0000gn/T/pip-wheel-ehajjkhy
       cwd: /private/var/folders/sh/xsq9cpw572d9fywbp9l0j35c0000gn/T/pip-install-fss2zby2/paho-mqtt/
  Complete output (54 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib
  creating build/lib/paho
  copying src/paho/__init__.py -> build/lib/paho
  creating build/lib/paho/mqtt
  copying src/paho/mqtt/__init__.py -> build/lib/paho/mqtt
  copying src/paho/mqtt/client.py -> build/lib/paho/mqtt
  copying src/paho/mqtt/matcher.py -> build/lib/paho/mqtt
  copying src/paho/mqtt/packettypes.py -> build/lib/paho/mqtt
  copying src/paho/mqtt/properties.py -> build/lib/paho/mqtt
  copying src/paho/mqtt/publish.py -> build/lib/paho/mqtt
  copying src/paho/mqtt/reasoncodes.py -> build/lib/paho/mqtt
  copying src/paho/mqtt/subscribe.py -> build/lib/paho/mqtt
  copying src/paho/mqtt/subscribeoptions.py -> build/lib/paho/mqtt
  running egg_info
  writing src/paho_mqtt.egg-info/PKG-INFO
  writing dependency_links to src/paho_mqtt.egg-info/dependency_links.txt
  writing requirements to src/paho_mqtt.egg-info/requires.txt
  writing top-level names to src/paho_mqtt.egg-info/top_level.txt
  reading manifest file 'src/paho_mqtt.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  writing manifest file 'src/paho_mqtt.egg-info/SOURCES.txt'
  installing to build/bdist.macosx-10.9-x86_64/wheel
  running install
  running install_lib
  creating build/bdist.macosx-10.9-x86_64
  creating build/bdist.macosx-10.9-x86_64/wheel
  creating build/bdist.macosx-10.9-x86_64/wheel/paho
  copying build/lib/paho/__init__.py -> build/bdist.macosx-10.9-x86_64/wheel/paho
  creating build/bdist.macosx-10.9-x86_64/wheel/paho/mqtt
  copying build/lib/paho/mqtt/__init__.py -> build/bdist.macosx-10.9-x86_64/wheel/paho/mqtt
  copying build/lib/paho/mqtt/client.py -> build/bdist.macosx-10.9-x86_64/wheel/paho/mqtt
  copying build/lib/paho/mqtt/matcher.py -> build/bdist.macosx-10.9-x86_64/wheel/paho/mqtt
  copying build/lib/paho/mqtt/packettypes.py -> build/bdist.macosx-10.9-x86_64/wheel/paho/mqtt
  copying build/lib/paho/mqtt/properties.py -> build/bdist.macosx-10.9-x86_64/wheel/paho/mqtt
  copying build/lib/paho/mqtt/publish.py -> build/bdist.macosx-10.9-x86_64/wheel/paho/mqtt
  copying build/lib/paho/mqtt/reasoncodes.py -> build/bdist.macosx-10.9-x86_64/wheel/paho/mqtt
  copying build/lib/paho/mqtt/subscribe.py -> build/bdist.macosx-10.9-x86_64/wheel/paho/mqtt
  copying build/lib/paho/mqtt/subscribeoptions.py -> build/bdist.macosx-10.9-x86_64/wheel/paho/mqtt
  running install_egg_info
  Copying src/paho_mqtt.egg-info to build/bdist.macosx-10.9-x86_64/wheel/paho_mqtt-1.5.0-py3.7.egg-info
  running install_scripts
  adding license file "LICENSE.txt" (matched pattern "LICEN[CS]E*")
  dyld: lazy symbol binding failed: Symbol not found: _fdopendir$INODE64
    Referenced from: /Users/perhuttner/anaconda3/envs/eegsynth/bin/python
    Expected in: /usr/lib/libSystem.B.dylib
  
  dyld: Symbol not found: _fdopendir$INODE64
    Referenced from: /Users/perhuttner/anaconda3/envs/eegsynth/bin/python
    Expected in: /usr/lib/libSystem.B.dylib
  
  ----------------------------------------
  ERROR: Failed building wheel for paho-mqtt
```

robertoostenveld, 2020-04-04:
The problem is that paho-mqtt needs something to be compiled, whith a library that is not supported on macOS 10.9.5. Info can be found when googling `dyld: Symbol not found: _fdopendir$INODE64`

I was able to work around the problem with
```
pip install --no-deps eegsynth
```

Subsequently, eegsynth was installed, but some dependencies were still missing. I struggled my way through them using either `pip install` or `conda install`. Right now I am still stuck on Qt not working.

stephenwhitmarsh, 2020-04-04:
kut

robertoostenveld, 2020-04-12:
upgrading to macOS 10.10 was enough to get it going again

closed 2020-04-12 by robertoostenveld