eegsynth/eegsynth #376: why: Warning: outputgpio only works on a Raspberry Pi

stephenwhitmarsh, 2020-04-12:
...when not running outputgpio. To remove confusion I'd rather opt for removing this warning alltogether if it proves complicated.

stephenwhitmarsh, 2020-04-12:
Same with: 
Warning: pandas is required for the cogito module, please install it with "pip install pandas"
Warning: neurokit is required for the complexity module, please install it with "pip install neurokit"

stephenwhitmarsh, 2020-04-12:
If we want people to take warnings seriously, we should not give them when they are not needed, re. boy shouting wolf!

robertoostenveld, 2020-04-12:
please look at the code of `eegsynth.py` and give a constructive suggestion on how to deal with modules (i.e. classes) that cannot be instantiated. Simply complaining does not solve the problem.  Perhaps thow an error of a specific type and have that error be caught in `eegsynth.py` or so.

closed 2020-04-12 by robertoostenveld

stephenwhitmarsh, 2020-04-12:
It's not a complaint, but jotting down of ideas for improvements to be
thought through at a later date. Just trying to keep them at a central
place. If not here, where then?