eegsynth/eegsynth #390: Installation warning (?) on Anaconda (Mac OS 10.14.6)

F13rr0, 2022-04-07:
Hello,

David Fierro here from the BBDMI project, together with Stephen. 

We just installed all the dependencies on Anaconda according to the general installation instructions. All seemed well until the final `pip install eegsynth`, where we got the following error regarding PyBluez-bitalino. However it does seem it was dealt with in the end (see message). In fact, do we really need the Bitalino library to be part of the EEGsynth installation?

Best,
David & Stephen

---
Building wheels for collected packages: PyBluez-bitalino, paho-mqtt, sklearn, termcolor
  Building wheel for PyBluez-bitalino (setup.py) ... error
  **ERROR: Command errored out with exit status 1:
   command: /opt/anaconda3/envs/eegsynth/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/b7/l29j2v5d5jl12_dtyg9j2ft00000gn/T/pip-install-ajzo4r41/pybluez-bitalino_3b657faa33934dc49270cfe1031053d6/setup.py'"'"'; __file__='"'"'/private/var/folders/b7/l29j2v5d5jl12_dtyg9j2ft00000gn/T/pip-install-ajzo4r41/pybluez-bitalino_3b657faa33934dc49270cfe1031053d6/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/b7/l29j2v5d5jl12_dtyg9j2ft00000gn/T/pip-wheel-7yr_7oxk
       cwd: /private/var/folders/b7/l29j2v5d5jl12_dtyg9j2ft00000gn/T/pip-install-ajzo4r41/pybluez-bitalino_3b657faa33934dc49270cfe1031053d6/**
  Complete output (8 lines):
  xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/private/var/folders/b7/l29j2v5d5jl12_dtyg9j2ft00000gn/T/pip-install-ajzo4r41/pybluez-bitalino_3b657faa33934dc49270cfe1031053d6/setup.py", line 105, in <module>
      "DEPLOYMENT_LOCATION=YES",
    File "/opt/anaconda3/envs/eegsynth/lib/python3.7/subprocess.py", line 363, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['xcodebuild', 'install', '-project', 'macos/LightAquaBlue/LightAquaBlue.xcodeproj', '-scheme', 'LightAquaBlue', 'DSTROOT=/private/var/folders/b7/l29j2v5d5jl12_dtyg9j2ft00000gn/T/pip-install-ajzo4r41/pybluez-bitalino_3b657faa33934dc49270cfe1031053d6/macos', 'INSTALL_PATH=/', 'DEPLOYMENT_LOCATION=YES']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for PyBluez-bitalino
  Running setup.py clean for PyBluez-bitalino
  Building wheel for paho-mqtt (setup.py) ... done
  Created wheel for paho-mqtt: filename=paho_mqtt-1.6.1-py3-none-any.whl size=62133 sha256=0f26cc9cfcd92f58d1d8cf2bfa223a7a1b59f79009ff310082a0c567d74e51de
  Stored in directory: /Users/davidfierro/Library/Caches/pip/wheels/d0/bf/ac/2b3f43f8c6fcd0f4ba5395397458c521eb0b52d33b574a5a40
  Building wheel for sklearn (setup.py) ... done
  Created wheel for sklearn: filename=sklearn-0.0-py2.py3-none-any.whl size=1309 sha256=7ffd6004a16d028d6889ccb86fcd663f27f9a863ca1089950cc5721e2b117a58
  Stored in directory: /Users/davidfierro/Library/Caches/pip/wheels/46/ef/c3/157e41f5ee1372d1be90b09f74f82b10e391eaacca8f22d33e
  Building wheel for termcolor (setup.py) ... done
  Created wheel for termcolor: filename=termcolor-1.1.0-py3-none-any.whl size=4847 sha256=acc510b1b9113fb6ee580771e0022e7dbd03819227d7fc2f3f452a5f86c2225b
  Stored in directory: /Users/davidfierro/Library/Caches/pip/wheels/3f/e3/ec/8a8336ff196023622fbcb36de0c5a5c218cbb24111d1d4c7f2
Successfully built paho-mqtt sklearn termcolor
Failed to build PyBluez-bitalino
Installing collected packages: pyobjc-core, pyobjc-framework-Cocoa, pyserial, PyBluez-bitalino, nibabel, fuzzywuzzy, termcolor, sklearn, python-osc, pylsl, paho-mqtt, nilearn, mido, configparser, bitalino, eegsynth
    Running setup.py install for PyBluez-bitalino ... done
  DEPRECATION: PyBluez-bitalino was installed using the legacy 'setup.py install' method, because a wheel could not be built for it. A possible replacement is to fix the wheel build issue reported above. You can find discussion regarding this at https://github.com/pypa/pip/issues/8368.
**Successfully installed PyBluez-bitalino-0.30 bitalino-1.2.6 configparser-5.2.0 eegsynth-0.4.2 fuzzywuzzy-0.18.0 mido-1.2.10 nibabel-3.2.2 nilearn-0.9.0 paho-mqtt-1.6.1 pylsl-1.15.0 pyobjc-core-8.4.1 pyobjc-framework-Cocoa-8.4.1 pyserial-3.5 python-osc-1.8.0 sklearn-0.0 termcolor-1.1.0**

robertoostenveld, 2022-04-07:
I think that the error relates to this https://github.com/nodejs/node-gyp/issues/569, i.e. Xtools is not yet installed.  

However, PyBluez-bitalino is only needed for the `bitalino2ft` module, so you should be fine if you are not going to use a Bitalino system.

stephenwhitmarsh, 2022-04-07:
> I think that the error relates to this nodejs/node-gyp#569

Thanks. We also just noticed at a later stage that Xtools that was
installed was too old, and for newer versions the macbook is too old. We
are trying to install the right version now. We will try again when/if that
succeeds and report back here.

closed 2022-04-11 by robertoostenveld