eegsynth/eegsynth #391: I don't understand the default filter frequencies

stephenwhitmarsh, 2022-04-20:
Hi @robertoostenveld,

In preprocessing module, filter default settings are set as follows (line 143->):

    ```
    try:
        float(config.get('processing', 'highpassfilter'))
        float(config.get('processing', 'lowpassfilter'))
        float(config.get('processing', 'notchfilter'))
        # the filter frequencies are specified as numbers
        default_scale = 1.
    except:
        # the filter frequencies are specified as Redis channels
        # scale them to the Nyquist frequency
        default_scale = hdr_input.fSample/2

    monitor.info('default scale for filter settings is %.0f' % (default_scale))

    scale_lowpass       = patch.getfloat('scale', 'lowpassfilter', default=default_scale)
    scale_highpass      = patch.getfloat('scale', 'highpassfilter', default=default_scale)
    scale_notchfilter   = patch.getfloat('scale', 'notchfilter', default=default_scale)
```

How I read this, is that any missing value for any of the filter settings (hp/lp/notch) will result in all filters being set to the upper cutoff. This would not be appropriate for the hp/notch, right? Should the first `try` and `default_scale` not be separate for the different filters?

stephenwhitmarsh, 2022-04-20:
Ah, I now see that this only concerns the scale/offset of the filter. Closed (but recorded for future laughts :-)

closed 2022-04-20 by stephenwhitmarsh

robertoostenveld, 2022-04-20:
the idea is that 
- when specified as numbers, those numbers are in Hz
- when specified as Redis channel, it is expected to contain values between 0 and 1 which are to be scaled to convert them to Hz

The default for the scaling is therefore 1 in the first situation, and Fn/2 (which is input data stream dependent) in the 2nd situation. Those defaults allow skipping the explicit specification of the scaling and offset in most cases.   

The default/example `preprocessing.ini` is not really demonstrating the smartness of the defaults in the code.

stephenwhitmarsh, 2022-04-20:
Actually, that IS very smart! :-) And actually very useful, especially in the case of EMG filtering before RMS (which I am implementing now)