eegsynth/eegsynth #393: Add on/off plotting option in plotsignal (from Redis)

stephenwhitmarsh, 2022-04-21:
I made a patch that includes EEG and EMG processing and plotting. It was extremely slow with processing on control signals ~10 times slower than required (perhaps even more). I identified the problem to be the signal plotting. They themselves do not lag at all, indicating PyQT gets some kind of priority over other processes. Closing them sped up the whole pipeline tremendously, and back to what was expected/requested of the other modules.
Plotting is important, however, when evaluating the signal. If we add the option to pause/unpause plotting, e.g. via a Redis value received via midi or inputcontrol, we can have the best of both worlds and switch to plotting priority or process priority. To further improve efficiency, I would suggest to pause the reading of the data from the FT buffer then as well, not just the plotting.
The plotting of the control signals does not seem to require extra undue slowing, by the way, and would not require such option at this point.

robertoostenveld, 2022-04-21:
If we are going to en/disable plotting, I suggest to do it consistently in all plotting functions. And I propose to model it after the en/disabling of the saving of data to disk (with the `recordxxx` functions). Having a consistent implementation for en/disabling a module will facilitate documentation and avoid inconsistent implementations.

stephenwhitmarsh, 2022-04-21:
True, and historycontrol (if not already consistent)

robertoostenveld, 2022-07-03:
We have some modules that already have an on/off option in one of the other sections in the ini file, but I think that the `[general]` section is the most appropriate for this.

I added the following option to a number of modules.

```
[general]
enable=1     ; this can be patched to Redis
```

and I have tested it with `inputcontrol` to switch plotting on/off.