eegsynth/eegsynth #420: Polarity of the signal from OpenBCI

faturita, 2023-05-17:
We are trying to implement an artistic solution using OpenBCI and have an issue with the polarity of the obtained signal

In https://github.com/eegsynth/eegsynth/blob/master/doc/openbci.md you mention that:

> Rumor has it that with this configuration, the output of the ADS1299 and hence the Cython board is inverted.

Do you have more information about this "Rumor" ?  Does this occur when the N pins are referenced to SRB2 (closer to the board) or to SRB1 (farther away from the board) ??   

Thanks in advance for any help !
PS: I know this is not an issue, or it could be to enhance the documentation :)

robertoostenveld, 2023-05-17:
In the documentation of the openbci2ft module (which I am just cleaning up, see #422) we had the comment 

> Warning in the [schematics](https://raw.githubusercontent.com/OpenBCI/Docs/master/assets/images/OBCI_V3_32bit-Schematic.jpg) of the OpenBCI cyton board the SRB1 and SBB2 labels are swapped; in the TI [ADS1299 datasheet](http://www.ti.com/lit/ds/symlink/ads1299.pdf) you can see that SRB1 is pin 17 on the corner, and hence following the [PCB layout](https://raw.githubusercontent.com/OpenBCI/Docs/master/assets/images/OBCI_32bit_layerTop.jpg) connected to the top pin of the double cyton header.

robertoostenveld, 2023-05-17:
from the https://www.ti.com/lit/ds/symlink/ads1299.pdf datasheet

[image]

robertoostenveld, 2023-05-17:
here is a cutout from the PCB layout, where the traces of pin 17 and 18 are followed (see the labels added to the figure) to the angled pin header.

[image]

Following the pin header that angles up, i.e. away from the AD1299 and toward the edge of the PCB, it follows that the ADS1299 pin18=SRB2 is the header pin close to the PCB board, and that  pin17=SRB1 is further away from the PCB board.

robertoostenveld, 2023-05-17:
Whereas in [this schematic](https://raw.githubusercontent.com/OpenBCI/Docs/master/assets/images/OBCI_V3_32bit-Schematic.jpg) the labels of pin 17 and 18 are marked incorrectly. 
 
[image]

robertoostenveld, 2023-05-17:
I don't think this addresses your issue yet, but I just wanted to have this more explicitly documented now that I have just deleted this snippet of documentation from the master branch of the eegsynth repository.

robertoostenveld, 2023-05-17:
Also, IN1P (positive) is the pin on the angled header that is further away from the PCB board and IN1N is the pin closest to the PCB board. 

With a bipolar configuration you would record IN1P minus IN1N on channel 1. With a unipolar configuration - and given figure 73 and page 68 in the ADS1299 datasheet - you would record IN1P minus SRB1 on channel 1. 

I don't recall where the pinout specification of OpenBCI can be found online, the documentation has moved around on https://docs.openbci.com and right now I cannot find the exact pinout. Of course there is also still the firmware on the PIC32 that can complicate matters, as well as the sign in the transmission protocol and in the software.

faturita, 2023-05-22:
We performed the Eyeball dipole experiment (e.g. https://github.com/PHYS3888/SpikerBox_EyeDipole) and we confirm that it is not a rumor, it is real, and that the signal that you get from OpenBCI using INxN - SRB2 (the pin closer to the board) is indeed INVERTED.

We connected EOG1 (1 cm left of left eye cantus and a little down) to the SRB2 and  EOG2 (1 cm above right eye center) to INxN (pin closer to the board) and we verified that center-right eyes movement produced a negative peak and center-left movement produced a positive peak (i.e. moving the POSITIVE CORNEA towards the electrode connected to INxN pin gives you a NEGATIVE deflection).

Thank you a lot for the information.

robertoostenveld, 2023-05-25:
Thanks for sharing. 

I tried to get my OpenBCI Cyton again to work to see for myself, but they would not give a stable connection and the signal would drop every time after 10-20 seconds irrespective of the distance between board and dongle. This applied to both Cytons that I have (I have a 8-channel and a 16-channel). Disappointing, but I guess that after not being used for a few years the solder connections on the PCB have deteriorated, resulting in the boards not being stable any more. Perhaps I could try to reflow the PCB solder connections... but  I was never happy with them anyway, so I don't think that I should invest further in them.