eegsynth/eegsynth #423: Adding support for BioAmp EXG Pill

lorforlinux, 2024-01-27:
It would be awesome to have support for https://github.com/upsidedownlabs/BioAmp-EXG-Pill 

I am willing to invest time to help develop this, please provide some guidance for the same.

robertoostenveld, 2024-01-27:
Hi Deepak, that would indeed be nice!

The ExG-pill is an analog board, which means that interfacing it first requires an AD conversion, and then some mechanism for digital interfacing with the computer running the eegsynth software. This compares to the approach for the https://pulsesensor.com, which we have used with the EEGsynth.

In the past I started off using this https://github.com/robertoostenveld/arduino/tree/main/PulseSensor_v2 (note there is also a v1 and a v0 version), which uses an arduino for ADC and a serial interface for communication. The data is communicated using the ancient OpenEEG serial format, for which we already had an interface https://www.fieldtriptoolbox.org/development/realtime/modulareeg/ to get it into the FieldTrip buffer. The disadvantage of that approach is that it requires a wired (serial) connection between arduino and computer, or a serial-over-bluetooth connection (which in my experience are not that stable, and has range issues). Furthermore, it requires a C/C++ compiled application to read from serial and write to the buffer.

Hence later I implemented an alternative https://github.com/robertoostenveld/arduino/tree/main/esp8266_pulse_sensor which connects the PulseSensor to an ESP8266 for the ADC, and which them sends the data over wifi, directly using the FieldTrip buffer interface. Using this approach, you don't need any special software on the eegsynth computer; it is sufficient to have the buffer running that receives the data.

If I recall correctly, https://github.com/robertoostenveld/arduino/tree/main/esp8266_fieldtrip_buffer is an even simpler demo or proof-of-principle application that similarly acts as a client and streams fake simulated data from the ESP8266 to the buffer server. I would start with this, get yourself an ESP development board, see whether the data features of the ExG-Pill (sampling rate and number of channels) can be accommodated by the ESP module and the wifi interface, and then hook the ESP up to the ExG-Pill.

Nowadays for new projects I would prefer an ESP32 over an ESP8266. I don't see why the code that I linked to above would not also run on an ESP32, but I have not tried it out.

best regards,
Robert

robertoostenveld, 2024-01-28:
I remembered that I have another example: with https://github.com/robertoostenveld/arduino/tree/main/esp8266_ad8232_ecg I connected an [AD8232 ECG Heart Monitor](https://www.sparkfun.com/products/12650) (which are available for cheap on AliExpress and Bay) to an ESP8266, also streaming the data directly over WiFi into the FieldTrip buffer and thereby the EEGsynth software. 

I apologize for it not being better documented. I still have the physical device in a small 3d printed enclosure somewhere on my attic, if needed I could take some photo's and look at how it again works in detail. Looking at the code in the ino file, I see it uses two blocks for buffering: while one block is being filled with values from the ADC using [Ticker](https://www.arduino.cc/reference/en/libraries/ticker/) and a callback every ms, the other block gets transmitted in the main loop. Now looking at the https://github.com/robertoostenveld/arduino/tree/main/esp8266_pulse_sensor ino code, I see it uses the same approach.

lorforlinux, 2024-01-28:
Hey Robert, thanks a lot for providing the details this is a huge help!
I have enough information now to work on this project, thanks again for your generous support. I'll post my development updates here. Very excited to get this up and running 💯

lorforlinux, 2024-01-29:
Hey Robert, we can surely send you an EXG kit with BioAmp EXG Pills. Thank you so much for all the information. I will connect with you via email for the shipment and further discussion on this project.