eegsynth/eegsynth #429: Values get set/published as "np.float64()" strings

caurnhammer, 2024-11-21:
I am getting nan values when running the historycontrol module due to values being set as “np.float64()” strings by the spectral module.

**Conda environment**
Python version 3.12
Numpy version 2.1.3

**Patch**
My patch is receiving EEG data via LSL from a Mentalab Explore Pro device and is extracting spectral features.
I am running the following modules:
- buffer
- preprocessing
- spectral
- historycontrol

This is caused by spectral setting and publishing values containing np.float64s as strings:

**Example logs**

redis-cli monitor
```
1732186160.088172 [0 127.0.0.1:57187] "PUBLISH" "spectral.channel6.theta" "np.float64(852.779177541534)"
1732186160.088369 [0 127.0.0.1:57187] "SET" "spectral.channel7.theta" "np.float64(754.9974777401796)"

1732186160.096902 [0 127.0.0.1:57193] "SET" "spectral.channel6.theta.mean" "np.float64(nan)"
1732186160.096959 [0 127.0.0.1:57193] "PUBLISH" "spectral.channel6.theta.mean" "np.float64(nan)"
```

spectral
```
DEBUG: spectral: ['delta', 'theta', 'alpha', 'beta', 'gamma'] [2.0, 5.0, 9.0, 14.625, 35.0] [5.0, 8.0, 11.25, 23.625, 45.0]
DEBUG: spectral: [389. 447. 531. 366. 590. 530. 614. 616. 439. 366. 338. 350. 385. 483.
 563. 501. 916. 380. 341. 301. 384. 416. 363. 374. 630. 319. 245. 194.
 259. 184. 180. 195. 579. 282. 182. 178. 197. 145. 149. 150.]
```

historycontrol
```
DEBUG: historycontrol: spectral.channel5.delta.min_att = nan
DEBUG: historycontrol: spectral.channel5.theta.min_att = nan
DEBUG: historycontrol: spectral.channel5.alpha.min_att = nan
```

**Workaround**

Practically, this can by fixed by the following change: 
spectral.py:ln 191   
`patch.setvalue(key, value[i])` ->  `patch.setvalue(key, float(value[i]))`

However, this does’t address the problem at its root and the same problem can pop up in a few other modules (e.g., ln. 146 in postprocessing). Do you know where and why this np.float64() string might be introduced in the first place?

robertoostenveld, 2024-11-21:
Hi Christoph

I recall a problem like this showing up earlier, and that it had to do with either a Python version update and/or a Redis update. The problem had to do with scalars either being "normal" Python scalars, or numpy objects. 

Note that this relates to #331, which is a long-standing open issue with no (or little) work devoted to it.

I'll look into it, see whether I can find the previous occurrence back and also learn something from that.

best regards,
Robert

robertoostenveld, 2024-11-21:
I think I can reproduce something that is related: With python 3.10.14 and numpy 1.26.4 (which happens to be my default "eegsynth" coda environment, already for a long time), I get this

```
>>> import numpy as np
>>> a = float('nan')
>>> b = np.float64('nan')
>>> type(a)
<class 'float'>
>>> type(b)
<class 'numpy.float64'>
>>> a.__repr__()
'nan'
>>> b.__repr__()
'nan'
```

Whereas with python 3.12 and numpy 2.1.3 (like yours) I get this

```
>>> import numpy as np
>>> a = float('nan')
>>> b = np.float64('nan')
>>> type(a)
<class 'float'>
>>> type(b)
<class 'numpy.float64'>
>>> a.__repr__()
'nan'
>>> b.__repr__()
'np.float64(nan)'
```

So it has to do how numbers are [represented](https://www.codecademy.com/resources/docs/python/dunder-methods/repr). I wonder whether this is only a problem for nans, or whether it would also be a problem for other scalars. With numpy 2.1.3 I also get 

```
>> c = np.float64(1.0)
>>> c.__repr__()
'np.float64(1.0)'
```

So I would expect that also to be problematic on the receiving end.

This is related https://stackoverflow.com/questions/78630047/how-to-stop-numpy-floats-being-displayed-as-np-float64 and points to this https://numpy.org/devdocs/release/2.0.0-notes.html#representation-of-numpy-scalars-changed. The [numpy.set_printoptions](https://numpy.org/devdocs/reference/generated/numpy.set_printoptions.html) may be a good solution. 

Pending further investigations, a workaround for you would be to set up a conda environment with numpy 1.x (like mine).

robertoostenveld, 2024-11-21:
Thanks for that feedback. So it is up to us to get it to work with Numpy 2.x ... to be continued

robertoostenveld, 2024-11-21:
I just noticed that in the `pyproject.toml` it is stated in the dependencies that `"numpy < 2.0.0"`, which I now recall as the work-around in the previous encounter of this nan/float64 problem. When one does `pip install eegsynth` (from pypi) then the right numpy would (or should) be installed. If one downloads the git repo in their own environment, that is not guaranteed of course.

caurnhammer, 2024-11-21:
I followed the [installation instructions](https://github.com/eegsynth/eegsynth/blob/master/doc/installation.md) again in a fresh conda environment.

Running `conda install numpy` results in my 2.x version, as nothing else is specified. 
Following this up by `pip install eegsynth` does not appear to downgrade numpy to < 2.0.0:

`Requirement already satisfied: numpy in /Users/christoph/Apps/miniconda3/envs/eegsynth_test/lib/python3.12/site-packages (from eegsynth) (2.1.3)`

robertoostenveld, 2024-11-23:
Ok, so the `<2.x` in the `pyproject.toml` apparently does not do what it is supposed to... That is another thing to look into. 

The code update that I just pushed prevents the `np.float(nan)` values in Redis, and appears to solve the problem for me. I hope you can test this once more and confirm.

closed 2024-11-23 by robertoostenveld

caurnhammer, 2024-11-23:
Everything works now: Pulled your changes and tested again with numpy 2.1.3.

Thanks, Robert!

commit 816e8e9e (2024-11-23): FIX #429 - convert numpy.float/int/bool values into normal Python scalars